PaulStoffregen/TimerOne #4: Not working on 1284P - have fix

bHogan2, 2015-05-09:
TimerOne is working on 328P AND the 644P but not on 1284P.
I am using the latest TimerOne and Mighty 1284 core.
Tried on IDE 1.0.6 and 1.6.4. with more than one chip.

I noticed a diff in pins between the 644 and 1284. 

#elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644__)
  #define TIMER1_A_PIN   13
  #define TIMER1_B_PIN   12
. . . versis . . .
#elif defined(__AVR_ATmega1284P__)
  #define TIMER1_A_PIN   12  // PD5 <<<<<< SB 13?
  #define TIMER1_B_PIN   13  // PD4 <<<<<<< SB 12?

I switched 12 and 13  for the 1284 and it's working now.
Hopefully it's a typo and I don't have some kind of bizarro problem :)

Thanks for a great lib and all the work to support so many devices.
John

PaulStoffregen, 2016-02-28:
Looks like this was fixed long ago.

    // Mighty-1284 - Maniacbug
    //
    #elif defined(__AVR_ATmega1284P__)
      #define TIMER1_A_PIN   12  // PD5
      #define TIMER1_B_PIN   13  // PD4
      #define TIMER1_ICP_PIN 14  // PD6
      #define TIMER1_CLK_PIN 1   // PB1
      #define TIMER3_A_PIN   6   // PB6
      #define TIMER3_B_PIN   7   // PB7
      #define TIMER3_ICP_PIN 5   // PB5
      #define TIMER3_CLK_PIN 8   // PD0

I'm closing this old issue.  If there's still a problem, please reopen and explain...

MadGarra, 2016-07-15:
Sorry, but it is not fixed.

vdesmedt, 2020-09-16:
I confirm the problem is not solved.

This code :
`void setup()
{
  //Timer1.initialize(1000);
  //Timer1.attachInterrupt(timerIsr);
  analogWrite(P_FAN_PWM, 64);
}`
Produces correct result

while this code 
`void setup()
{
  Timer1.initialize(1000);
  Timer1.attachInterrupt(timerIsr);
  analogWrite(P_FAN_PWM, 64);
}`

Creates a very thin spike instead of the square wave the above code produces

eeproto, 2021-03-25:
I just ran into the same issue. Things work fine on the ATmega1284 when you modify `known_16bit_timers.h` to read:

```CPP
// Mighty-1284 - Maniacbug
//
#elif defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega1284__)
  #define TIMER1_A_PIN   13  // PD5
  #define TIMER1_B_PIN   12  // PD4
  #define TIMER1_ICP_PIN 14  // PD6
  #define TIMER1_CLK_PIN 1   // PB1
  #define TIMER3_A_PIN   6   // PB6
  #define TIMER3_B_PIN   7   // PB7
  #define TIMER3_ICP_PIN 5   // PB5
  #define TIMER3_CLK_PIN 8   // PD0

#endif
```

This also enables TimerOne to run on a non-P ATmega1284. Tested with an ATmega1284-MU.