PaulStoffregen/TimerOne #8: Doesn't support non-avr boards?

Gargo, 2016-02-01:
Wemos R1 D1 + this example:
http://www.lucidarme.me/?p=5057

Or could you help at least how to change the code from this example to avoid to use TimerOne?

PaulStoffregen, 2016-02-01:
TimerOne does work with Arduino Mega

Gargo, 2016-02-02:
Wemos R1 D1 is not Mega. It is almost Uno clone based on ch340g + built-in esp8266. The only difference from Uno I saw is it has only one analog pin

PaulStoffregen, 2016-02-02:
Well, if you or someone else writes the code, I'll be happy to merge it.  But I do not have this board, and even if I did, I wouldn't have the time available to write this myself.

closed 2016-02-02 by PaulStoffregen

Gargo, 2016-02-02:
Ok about time. But you should buy it because it is so cheap that you can buy one for each shield you have

PaulStoffregen, 2016-02-02:
Maybe you should ask the people who make and support this product to contribute code to this and other widely used libraries?