PaulStoffregen/TimerOne #10: timer time is doubled on arduino MEGA R3

aviad, 2016-05-17:
I might be missing something here, but it seems pretty straight forward:
I modified the file `examples/Interrupt/Interrupt.pde` to include a print of the time with `micros()`:
in the function `loop` I added 
```c_cpp
Serial.println("blinkCount = " + String(blinkCopy) + " @ " + String(micros()) + "us");
```

and changed the initialization to 500000 us to make it easier to follow, with `Timer1.initialize(500000);` in `setup`.

I'm using the arduino IDE on mac, just to be sure this is not a config issue.

Gists:
[code](https://gist.github.com/aviad/d5d5a7470443a5d586bc896a19d4dcfb)
[output](https://gist.github.com/aviad/c50eb673b3db40d1054afbf154b71fcc)

Example (Serial) output: the first blink takes 0.5 seconds, the rest 1.0 seconds.
```
blinkCount = 0 @ 60us
blinkCount = 0 @ 50516us
blinkCount = 0 @ 101064us
blinkCount = 0 @ 151652us
blinkCount = 0 @ 202236us
blinkCount = 0 @ 252828us
blinkCount = 0 @ 303416us
blinkCount = 0 @ 354004us
blinkCount = 0 @ 404596us
blinkCount = 0 @ 455176us
blinkCount = 1 @ 505768us
blinkCount = 1 @ 556360us
blinkCount = 1 @ 606944us
blinkCount = 1 @ 657532us
blinkCount = 1 @ 708120us
blinkCount = 1 @ 758708us
blinkCount = 1 @ 809300us
blinkCount = 1 @ 859884us
blinkCount = 1 @ 910472us
blinkCount = 1 @ 961056us
blinkCount = 1 @ 1011652us
blinkCount = 1 @ 1062272us
blinkCount = 1 @ 1112892us
blinkCount = 1 @ 1163512us
blinkCount = 1 @ 1214128us
blinkCount = 1 @ 1264752us
blinkCount = 1 @ 1315368us
blinkCount = 1 @ 1365992us
blinkCount = 1 @ 1416612us
blinkCount = 1 @ 1467236us
blinkCount = 2 @ 1517864us
blinkCount = 2 @ 1568480us
blinkCount = 2 @ 1619108us
blinkCount = 2 @ 1669728us
blinkCount = 2 @ 1720352us
blinkCount = 2 @ 1770968us
blinkCount = 2 @ 1821592us
blinkCount = 2 @ 1872220us
blinkCount = 2 @ 1922840us
blinkCount = 2 @ 1973460us
blinkCount = 2 @ 2024076us
blinkCount = 2 @ 2074700us
blinkCount = 2 @ 2125316us
blinkCount = 2 @ 2175940us
blinkCount = 2 @ 2226564us
blinkCount = 2 @ 2277180us
blinkCount = 2 @ 2327804us
blinkCount = 2 @ 2378424us
blinkCount = 2 @ 2429048us
blinkCount = 2 @ 2479668us
blinkCount = 3 @ 2530292us
blinkCount = 3 @ 2580916us
blinkCount = 3 @ 2631544us
blinkCount = 3 @ 2682164us
blinkCount = 3 @ 2732780us
blinkCount = 3 @ 2783404us
blinkCount = 3 @ 2834020us
blinkCount = 3 @ 2884644us
blinkCount = 3 @ 2935264us
blinkCount = 3 @ 2985888us
blinkCount = 3 @ 3036512us
blinkCount = 3 @ 3087128us
blinkCount = 3 @ 3137752us
blinkCount = 3 @ 3188368us
blinkCount = 3 @ 3238992us
blinkCount = 3 @ 3289608us
blinkCount = 3 @ 3340228us
blinkCount = 3 @ 3390852us
blinkCount = 3 @ 3441468us
blinkCount = 3 @ 3492096us
blinkCount = 4 @ 3542712us
blinkCount = 4 @ 3593336us
blinkCount = 4 @ 3643952us
blinkCount = 4 @ 3694576us
blinkCount = 4 @ 3745200us
blinkCount = 4 @ 3795824us
blinkCount = 4 @ 3846448us
blinkCount = 4 @ 3897064us
blinkCount = 4 @ 3947688us
blinkCount = 4 @ 3998304us
blinkCount = 4 @ 4048928us
blinkCount = 4 @ 4099544us
blinkCount = 4 @ 4150164us
blinkCount = 4 @ 4200788us
blinkCount = 4 @ 4251404us
blinkCount = 4 @ 4302032us
blinkCount = 4 @ 4352648us
blinkCount = 4 @ 4403272us
blinkCount = 4 @ 4453892us
blinkCount = 5 @ 4504516us
blinkCount = 5 @ 4555144us
blinkCount = 5 @ 4605760us
blinkCount = 5 @ 4656384us
blinkCount = 5 @ 4707000us
blinkCount = 5 @ 4757624us
blinkCount = 5 @ 4808240us
blinkCount = 5 @ 4858864us
blinkCount = 5 @ 4909488us
blinkCount = 5 @ 4960112us
blinkCount = 5 @ 5010740us
blinkCount = 5 @ 5061356us
blinkCount = 5 @ 5111980us
blinkCount = 5 @ 5162596us
blinkCount = 5 @ 5213220us
blinkCount = 5 @ 5263836us
blinkCount = 5 @ 5314460us
blinkCount = 5 @ 5365084us
blinkCount = 5 @ 5415704us
blinkCount = 5 @ 5466328us
blinkCount = 6 @ 5516948us
```

aviad, 2016-05-21:
@PaulStoffregen ?

PaulStoffregen, 2016-05-24:
I just returned from Maker Faire, and I have another event next weekend.  Can't look at this until June.

aviad, 2016-05-24:
thank you for updating. good luck.

lxmonk, 2016-05-31:
I just found out this was not an original Arduino. Can anyone verify this with an original, please?

arrow-, 2016-06-13:
I think you have misunderstood the program, `blinkCount` is incremented only when the LED turns ON, not in every interrupt.
The LED _state_ is toggled at each ISR, so it takes _two ISRs_ to turn it on.

| time | `blinkCount` | LED |
| ----- | ---------- | ---- |
| 0us | 0 | off |
| 500us | 1 | on |
| 1000us | 1 | off |
| 1500us | 2 | on |

aviad, 2016-06-13:
got it. thank you!

closed 2016-06-13 by aviad