PaulStoffregen/TimerOne #20: Old version of this library is in PlatformIO Library registry

Roman3349, 2016-11-28:
Hi!

PlatformIO library registry contains an old version of this library without Teensy 3.5 / 3.6 support.

Error: https://travis-ci.org/iqrfsdk/clibdpa-mcu/jobs/179476641#L1106

PaulStoffregen, 2016-11-28:
@ivankravets

ivankravets, 2016-11-28:
What do you mean under "old version"? @PaulStoffregen controls the versions of this library. The latest version is `1.1`. See http://platformio.org/lib/show/131/TimerOne/manifest

If your source code depends on the specific commit of this repo, you can use GIT URl with PlatformIO LIbrary Manager. See http://docs.platformio.org/en/latest/librarymanager/index.html

```
pio lib install https://github.com/PaulStoffregen/TimerOne.git
```

or in Project Configuration file
```ini
[env:mybuildenv]
board = teensy31
lib_deps = https://github.com/PaulStoffregen/TimerOne.git
```

P.S: I see that this repo contains 2 manifests: PIO + Arduino. Howerver, Arduino manifest is used. This library will be updated when new release will be created and `version` field will be modified.

Roman3349, 2017-01-27:
@PaulStoffregen Please release a new version of library with Teensy 3.6 support.