PaulStoffregen/TimerOne #39: Errors when using TimerOne with new Arduino Nano Every

rcmolina, 2019-08-19:
### Description

I get some errors, so some definitions may need to be added ¿? to support new ATMEGA4809.

### Steps To Reproduce Problem

Code compiling 

### Hardware & Software

Board Arduino Nano Every
Shields / modules used
Arduino IDE version 1.8.4
Teensyduino version (if using Teensy)
Version info & package name (from Tools > Boards > Board Manager)
Operating system & version win 8.1
Any other software or hardware?

### Arduino Sketch

```cpp
// Change the code below by your sketch (please try to give the smallest code which demonstrates the problem)
#include <Arduino.h>
#include <TimerOne.h>
// libraries: give links/details so anyone can compile your code for the same result

void setup() {
}

void loop() {
}
    Timer1.initialize(1000);                
    Timer1.attachInterrupt(wave);

### Errors or Incorrect Output
C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h: In member function 'void TimerOne::initialize(long unsigned int)':

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:47:2: error: 'TCCR1B' was not declared in this scope

  TCCR1B = _BV(WGM13);        // set mode as phase and frequency correct pwm, stop the timer

  ^~~~~~

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:47:2: note: suggested alternative: 'TCB1'

  TCCR1B = _BV(WGM13);        // set mode as phase and frequency correct pwm, stop the timer

  ^~~~~~

  TCB1

In file included from c:\users\rafael\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\io.h:99:0,

                 from c:\users\rafael\appdata\local\arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5\avr\include\avr\pgmspace.h:90,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/api/String.h:30,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/api/Print.h:24,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/api/Stream.h:25,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/api/Client.h:22,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/api/ArduinoAPI.h:29,

                 from C:\Users\Rafael\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.3\cores\arduino/Arduino.h:23,

                 from sketch\MaxDuino_v1.51.ino.cpp:1:

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:47:15: error: 'WGM13' was not declared in this scope
[... 497 lines cut ...]
                 from sketch\MaxDuino_v1.51.ino.cpp:1:

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:143:15: error: 'TOIE1' was not declared in this scope

  TIMSK1 = _BV(TOIE1);

               ^

In file included from C:\Users\Rafael\Documents\Arduino\MaxDuino_v1.51\MaxDuino_v1.51.ino:103:0:

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h: In member function 'void TimerOne::detachInterrupt()':

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:150:2: error: 'TIMSK1' was not declared in this scope

  TIMSK1 = 0;

  ^~~~~~

C:\Users\Rafael\Documents\Arduino\libraries\TimerOne/TimerOne.h:150:2: note: suggested alternative: 'TIMERB1'

  TIMSK1 = 0;

  ^~~~~~

  TIMERB1

jwalk487, 2019-08-19:
Same problem and struggling to find a workaround.

kryptonn, 2019-08-28:
Same here, even with REGISTER EMULATION ATMEGA328 enabled
It seems that nano every registers are completely different
Datasheet link : 
http://ww1.microchip.com/downloads/en/DeviceDoc/Atmega809-1609-3209-4809-48-Pin-40002016C.pdf

Kees-van-der-Oord, 2020-03-22:
Dear all,

I faced the same challenge as you and wrote a library for the MegaAVR TCB timer:
[https://github.com/Kees-van-der-Oord/Arduino-Nano-Every-Timer-Controller-B](https://github.com/Kees-van-der-Oord/Arduino-Nano-Every-Timer-Controller-B).

It has the same interface as the TimerOne library, so it should be easy to write code that works both on the AVR and MegaAVR chips. 

Please let me know if this work for you !

WR,
Kees