PaulStoffregen/TimerOne #46: probably flaw of start()

5chufti, 2020-11-21:
### Description
At least on UNO with latest Arduino IDE and lib git version
timer.start() probably not working as seen in sketches developed with older version.
after a timer.stop() a timer.start() immediately calls ISR (not delayed by set interval)

(maybe it is just an undocumented sideeffect of recent version that stop() clears the set interval?)

### Steps To Reproduce Problem

modified example shows problem:  see below

### Hardware & Software

Board UNO
Arduino IDE version
Version info & package name (from Tools > Boards > Board Manager) git version as of 21.Nov.2020
Operating system & version Win7

### Arduino Sketch

```cpp
#include <TimerOne.h>

const int led = LED_BUILTIN;  // the pin with a LED

void setup(void)
{
  pinMode(led, OUTPUT);
  Timer1.initialize(150000);
  Timer1.attachInterrupt(blinkLED); // blinkLED to run every 0.15 seconds
  Serial.begin(9600);
}

int ledState = LOW;
volatile unsigned long blinkCount = 0; // use volatile for shared variables

void blinkLED(void)
{
  Timer1.stop();
  if (ledState == LOW) {
    ledState = HIGH;
    blinkCount = blinkCount + 1;  // increase when LED turns on
  } else {
    ledState = LOW;
  }
  digitalWrite(led, ledState);
  Timer1.start();        //Timer1.initialize(150000); works in sketches w older version of lib
}

void loop(void)
{
  unsigned long blinkCopy;  // holds a copy of the blinkCount
  noInterrupts();
  blinkCopy = blinkCount;
  interrupts();

  Serial.print("blinkCount = ");
  Serial.println(blinkCopy);
  delay(100);
}
```

### Errors or Incorrect Output

not blinking, counter increasing much too fast w/o blink

vjenne, 2020-12-19:
Yes,  I just ran into this problem too. My solution is as follows: In timerOne.h, i changed line 85.
TCNT1 = 0;
I changed 0 to 1. I think preloading this timer generates an interrupt the same as when an overflow happens like  when the timer goed from 0xFFFF to 0x0000. Thus generating an overflow interrupt.
So if you change line 85 to 
TCNT1 = 1;
No interrupt will happen. Your timing will be 1 clock tick faster than expected....

jbidinger, 2022-09-03:
I changed the Timer1.start() call to Timer1.resume() in blinkLED() and your sketch seems to be working. I'm testing on a Mega though. Having said that I did have problems with Timer1.start().