PaulStoffregen/TimerOne #60: Compile Error with Updated A-Star 328PB Board Info

RossAWaddell, 2022-09-08:
I have two machines each with Arduino IDE v1.8.13 but only on one did I allow it to update the board info (I was prompted to so when starting up the Arduino IDE) and I now get a compile error with a very simple sketch:

**Sketch**
```
#include <TimerOne.h>           // https://github.com/PaulStoffregen/TimerOne

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

**Error message**:
```
Compiling sketch...
/Users/rwaddell/Library/Arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_A_STAR_328PB -DARDUINO_ARCH_AVR -DTWBR=TWBR1 -DTWSR=TWSR1 -DTWAR=TWAR1 -DTWDR=TWDR1 -DTWCR=TWCR1 -DTWAMR=TWAMR1 -DTWI_vect=TWI1_vect -DPIN_WIRE_SDA=PIN_WIRE_SDA1 -DPIN_WIRE_SCL=PIN_WIRE_SCL1 -DSPCR=SPCR1 -DSPSR=SPSR1 -DSPDR=SPDR1 -DPIN_SPI_SS=PIN_SPI_SS1 -DPIN_SPI_MOSI=PIN_SPI_MOSI1 -DPIN_SPI_MISO=PIN_SPI_MISO1 -DPIN_SPI_SCK=PIN_SPI_SCK1 -I/Users/rwaddell/Library/Arduino15/packages/arduino/hardware/avr/1.8.5/cores/arduino -I/Users/rwaddell/Library/Arduino15/packages/pololu-a-star/hardware/avr/5.1.0/variants/a-star328pb "-I/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne" /var/folders/nb/2356s9gj6y714v9fzf4hck080000gn/T/arduino_build_409503/sketch/sketch_sep08a.ino.cpp -o /var/folders/nb/2356s9gj6y714v9fzf4hck080000gn/T/arduino_build_409503/sketch/sketch_sep08a.ino.cpp.o
In file included from /Users/rwaddell/Dropbox/Arduino Projects/sketch_sep08a/sketch_sep08a.ino:1:0:
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h: In member function 'void TimerOne::setPwmDuty(char, unsigned int)':
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:244:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) OCR1A = dutyCycle;
             ^~~~~~~~~~~~
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:244:13: note: suggested alternative: 'TIMER1A'
  if (pin == TIMER1_A_PIN) OCR1A = dutyCycle;
             ^~~~~~~~~~~~
             TIMER1A
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h: In member function 'void TimerOne::pwm(char, unsigned int)':
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:253:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) { pinMode(TIMER1_A_PIN, OUTPUT); TCCR1A |= _BV(COM1A1); }
             ^~~~~~~~~~~~
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:253:13: note: suggested alternative: 'TIMER1A'
  if (pin == TIMER1_A_PIN) { pinMode(TIMER1_A_PIN, OUTPUT); TCCR1A |= _BV(COM1A1); }
             ^~~~~~~~~~~~
             TIMER1A
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h: In member function 'void TimerOne::disablePwm(char)':
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:268:13: error: 'TIMER1_A_PIN' was not declared in this scope
  if (pin == TIMER1_A_PIN) TCCR1A &= ~_BV(COM1A1);
             ^~~~~~~~~~~~
/Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne/TimerOne.h:268:13: note: suggested alternative: 'TIMER1A'
  if (pin == TIMER1_A_PIN) TCCR1A &= ~_BV(COM1A1);
             ^~~~~~~~~~~~
             TIMER1A
Using library TimerOne at version 1.1 in folder: /Users/rwaddell/Dropbox/Arduino Projects/libraries/TimerOne 
exit status 1
Error compiling for board Pololu A-Star 328PB.
```

AndresBefaco, 2022-10-20:
Same error here with an atmega 168PA

PaulStoffregen, 2022-10-25:
@RossAWaddell , @AndresBefaco - Please tell me which board you have selected in the Tools > Boards menu.  If you added a custom URL in File > Preferences and then used Boards Manager to install a package which provides that board, I need those details so I can install the same thing here, so I can reproduce the problem.

RossAWaddell, 2022-10-25:
@PaulStoffregen I was able to resolve the compile issue based on info from the Pololu forum:

```
The latest version of our A-Star board configuration no longer configures GCC to compile for the ATmega328P, but instead configures it to compile for the ATmega328PB. Unfortunately, if you try to install TimerOne using the library manager, you will get an older version (1.1) which does not support the ATmega328PB.

The latest version of the TimerOne library available on GitHub does seem to support the ATmega328PB. You can download it and copy it manually into your Arduino libraries folder to use it.
```

I manually downloaded the latest version of your library and that resolved the compile error.

But to answer your questions:

1. A-Star 328PB (5V, 16Mhz)
2. Arduino IDE Preferences: https://files.pololu.com/arduino/package_pololu_index.json (Additional Boards Manager URLs)

I don't think I installed a package to add the Pololu A-Star boards, but they do now show up separately under Tools > Boards

AndresBefaco, 2022-10-25:
@PaulStoffregen I used MiniCore by MCUDude: https://github.com/MCUdude/MiniCore 
Board168PA 16mhz external clock 
We also has been trying the 328PB for another project using the TimerOne library with the same compile error. I will see if this is  a library version issue as  was the case for @RossAWaddell

AndresBefaco, 2022-10-25:
Just tried the library update and now it compile with no issue, both the atmega 168PA and 328PB.
Thanks